sigmf/sigmf-python #6: Signal extension usage

AdrFebles, 2023-02-16:
Hi, 
I'm using SigMFFile to store metadata in the correct format. However I need to define more field for signal details using the "signal extension". I haven't find any Python module to associate in SigMFFile.EXTENSION_KEY to the signal extension described on the [SigMF repository](https://github.com/sigmf/SigMF/blob/sigmf-v1.x/extensions/signal.sigmf-ext.md). Are you planning to add this functionality in the SigMF Python package or do you have any idea about how can I add this info to the metadata?

Thanks in advance!

jacobagilbert, 2023-02-16:
Hi @AdrFebles - the `SigMFFile.EXTENSIONS_KEY` references the `extensions` field in the [`global` metadata object](https://github.com/sigmf/SigMF/blob/sigmf-v1.x/sigmf-spec.md#global-object) (or `collections` file). In general this module implements only the `core` spec and does not include any extensions, though it does not preclude manual references to any of the objects in extensions namespaces. There isn't really any overhead in directly using those objects as opposed to using `sigmffile` aliases other than they wont be directly validated (as they are not included in the schema files) if using the `validate()` function.

amulamoottil, 2023-06-05:
@jacobagilbert ,
So are you saying if I wanted to add Signal bandwidth in sigmf-meta file and wanted to get the bandwidth using sigmf-python file, I will not be able to as that functionality does not exist?

gregparkes, 2023-10-26:
@amulamoottil As far as I am aware, SigMF is fairly inflexible when it comes to adding additional meta information to your signals. The extensions module is pretty unwieldly and unintuitive to use.

I have personally taken the habit of using the `core:label` attribute of the Annotations to supply additional labelled information in my signals - parsing and unparsing using JSON, for example:

```python
import json
extra_labels = {"SNR": 30, "Fading": "Rician"}
my_annotation['core:label'] = json.dumps(extra_labels)
```

And for retrieving:

```python
for a in sig_handle.get_annotations():
    extra_labels = json.loads(a['core:label']) # returns a dict
```

However I believe there may be a character limit on this attribute, so pick one that works.

Alternatively, you can sidestep `sigmf` and simply load the .sigmf-meta file using JSON:

```python
with open("my_sigmf_file.sigmf-meta", "wt") as fin:
    meta_data = json.load(fin)
    meta_data['annotations']
```

That way you can add whatever labels you want (although obviously doesn't conform to Sigmf standards.)

Hope this helps.

closed 2024-12-20 by Teque5