sigmf/sigmf-python #7: SigMF webapp GUI is abandoned: adoption necessary or removal upcoming

jacobagilbert, 2023-02-20:
The SigMF GUI [is broken](https://github.com/sigmf/SigMF/issues/242), poorly documented, and has some [design issues](https://github.com/sigmf/SigMF/issues/199). It has not seen any work in over 6 years, which is leading me to consider removing it form the project.

I am seeking someone to adopt the GUI, fix its issues, and document its usage, or it will likely be removed. If we can get some traction here, we can also integrate this into the python module fully or possibly move it to a separate effort.

Teque5, 2023-09-01:
Yea if we don't find anyone at the 2023 GNU Radio Conference that's interested in maintaining I will remove it.

Teque5, 2023-12-20:
Closed with #38

closed 2023-12-20 by Teque5