sigmf/sigmf-python #11: Unable to create an archive with multiple recordings

jhazentia, 2023-04-19:
There does not appear to be a way to create an archive from a collection or with multiple recordings.

I am planning to work on this and submit a PR.

jacobagilbert, 2023-04-19:
Hi @jhazentia thanks for getting involved here.

@gmabey as our resident archive poweruser, do you have any insights for this?

gmabey, 2023-04-27:
Sorry for the delay in responding.

It definitely looks like you're right, that it's not supported anywhere right now.

However, if you're going to work on it, I suggest you work along the lines of `.add()`ing individual files instead of copying everything to a temporary directory and adding that directory.

If `fileobj is not None`, then looks like a `.addfile()` can be used to prevent writing the data to disk more than once.

Please do post progress and questions in this issue.