sigmf/sigmf-python #15: sigmffile.read_samples() ignores raw_components parameter

margamc, 2023-05-17:
Hi, I have detected what I think is a bug. When you do:
```
import sigmf
sigfile = sigmf.sigmffile.fromfile("some_sigmf_file")
samples_raw = sigfile.read_samples(raw_components=True)
samples = sigfile.read_samples()
```
The `samples_raw` and `samples` variables are identical. 

I have noticed that the `raw_components` parameter is not really used as argument for the internal `_read_datafile()` function in sigmffile.py. Instead, there is a static `False` when the function is called in line 605.

I have resolved this bug for my project by changing line 605 of sigmffile.py from:
```
return self._read_datafile(first_byte, count * self.get_num_channels(), autoscale, False)
```
to:
```
return self._read_datafile(first_byte, count * self.get_num_channels(), autoscale, raw_components)
```

But I don't know if this change has side effects. Please, could you check it and if it's ok solve the issue for the next release?

Thank you!

Teque5, 2025-09-21:
I was going to fix this issue and write a test, but in what occasion would we want raw components?

As I read it we are requesting samples neither interleaved for complex nor interleaved for channels. How could that possibly be useful?

@margamc are you using this functionality? It appears a PR from my org added this back in 2019, but it has no tests and I need clarification on why you would want it.

I think we should remove the ability to read raw components.

Teque5, 2026-01-21:
`raw_components` removed in PR #121 with `v1.4.0`.