sigmf/sigmf-python #17: unable to validate with python3.6

gmabey, 2023-06-07:
Attempting to use sigmf on RockyLinux 8.8 (python 3.6, jsonschema 2.6.0):
```python
[glenm@localhost sigmf-python]$ ipython
Python 3.6.8 (default, Jan 25 2023, 08:28:52) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.16.3 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import sigmf                                                                                                                                                       

In [2]: sigmf.SigMFArchiveReader('/host_tmp/2020-01-02_00-00-00.000.sigmf')                                                                                                
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-2-1b4bb3c06335> in <module>
----> 1 sigmf.SigMFArchiveReader('/host_tmp/2020-01-02_00-00-00.000.sigmf')

~/src/sigmf-python/sigmf/archivereader.py in __init__(self, name, skip_checksum, map_readonly, archive_buffer)
     73 
     74         self.sigmffile = SigMFFile(metadata=json_contents)
---> 75         valid_md = self.sigmffile.validate()
     76 
     77         self.sigmffile.set_data_file(self.name, data_buffer=archive_buffer, skip_checksum=skip_checksum, offset=data_offset_size[0],

~/src/sigmf-python/sigmf/sigmffile.py in validate(self)
    510         """
    511         version = self.get_global_field(self.VERSION_KEY)
--> 512         validate.validate(self._metadata, self.get_schema())
    513 
    514     def archive(self, name=None, fileobj=None):

~/src/sigmf-python/sigmf/validate.py in validate(metadata, ref_schema)
     66     None, will raise error if invalid.
     67     '''
---> 68     validator = jsonschema.Draft7Validator(schema=ref_schema)
     69     validator.validate(instance=metadata)
     70 

AttributeError: module 'jsonschema' has no attribute 'Draft7Validator'
```

gmabey, 2023-07-17:
resolved

closed 2023-07-17 by gmabey

gmabey, 2023-08-04:
@Teque5 dunno if you've seen the chat traffic, but I think you're the man to cut a python release, no?

Teque5, 2023-08-30:
Yes I will tag the repo and create a release and push a package to pip later today.