sigmf/sigmf-python #19: exception when calling get_default_metadata()

gmabey, 2023-06-08:
I don't use `sigmf.validate.get_default_metadata()` but it seems to be broken, in case someone *wanted* to use it:

```python
glenm@glenm-OptiPlex-7070:~/src/sigmf-python$ ipython3 
Python 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.31.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import sigmf

In [2]: sigmf.validate.get_default_metadata()
---------------------------------------------------------------------------
ValidationError                           Traceback (most recent call last)
<ipython-input-2-16c967c76391> in <module>
----> 1 sigmf.validate.get_default_metadata()

~/src/sigmf-python/sigmf/validate.py in get_default_metadata(ref_schema)
     44     default = {}
     45     validator = extend_with_default(jsonschema.Draft7Validator)
---> 46     validator(ref_schema).validate(default)
     47     return default
     48 

/usr/lib/python3/dist-packages/jsonschema/validators.py in validate(self, *args, **kwargs)
    351         def validate(self, *args, **kwargs):
    352             for error in self.iter_errors(*args, **kwargs):
--> 353                 raise error
    354 
    355         def is_type(self, instance, type):

ValidationError: 'global' is a required property
```
or maybe I'm missing something here?

Teque5, 2023-06-08:
It says in the docstring for that function that it is currently broken. I believe the intent was to retrieve a _vanilla_ metadata dictionary similar to how calling `sigmf.SigMFFile()` returns a basic dictionary:
```python
SigMFFile({
    "global": {
        "core:num_channels": 1,
        "core:version": "1.1.1"
    },
    "captures": [],
    "annotations": []
})
```
In the validator the empty default should get filled with a more realistic default. Currently this skeleton fails validation b/c it has no datatype defined.

Perhaps `get_default_metadata()` should be removed.

Teque5, 2024-09-10:
New PR removes this function and closes issue.

PR #69 (2024-09-10): delete concept of default metadata and clarify validation err
Closes #19.
closed 2024-09-10 by Teque5