sigmf/sigmf-python #20: Repo moved and README.md links are incorrect

mrdvt92, 2023-06-20:
On https://github.com/sigmf/sigmf-python, the Readme.md links are broken due to repo split and re-origination.

- link to https://github.com/sigmf/sigmf-python/blob/main/sigmf-spec.md is now a 404 error
- url https://github.com/gnuradio/SigMF is now redirected (I assume you want https://github.com/sigmf/SigMF/blob/HEAD/sigmf-spec.md)

```
The [SigMF specification document](https://github.com/sigmf/sigmf-python/blob/main/sigmf-spec.md), sigmf-spec.md is located in the https://github.com/gnuradio/SigMF repository.
```

Teque5, 2023-06-20:
Yea I noticed this a bit ago. We should also decide the policy on versioning and tag the repository per some other issues for prior and current versions. There were a few oversights from the move.

Teque5, 2023-09-01:
Fixed in PR https://github.com/sigmf/sigmf-python/pull/22.

closed 2023-09-01 by Teque5