sigmf/sigmf-python #23: Support converters for Wav & ?

Teque5, 2023-09-06:
A long time ago we decided against doing this when the repos all lived in SigMF, but we should re-assess.

At GRCon23 we discussed adding converters for common formats (not metadata formats like issue #14) to include `wav` and other common time-series formats.

777arc, 2023-09-08:
Yep, IQEngine was planning to add converters, so they could either live in this repo or in the IQEngine repo.  It would be all python-based converters and IQEngine would provide a page where someone can upload their non-sigmf recording and it will download the sigmf-meta.  Then for folks with sensitive recordings they can simply run the python converters manually.

Teque5, 2023-09-08:
In my brain this will work like this(?):
```python
# x is abc.wav, xyz.mp3, uvw.opus
import sigmf
handle = sigmf.sigmffile.fromfile(x)
# handle now has interpreted metadata from the file
samples = handle.read_samples()
# you can write this to sigmf archive or whatever
handle.toarchive('abc.sigmf')
```

If this looks good I could work up a demo pretty fast. I'm still unsure if we want to house such a converter in this repo, but it won't be that much code. There may be some lookup tables or other cumbersome things needed for the converters, but it might not be too bad.

777arc, 2023-09-08:
Ideally there would be a python script someone can run, providing their original format as input and it generates the SigMF files

Teque5, 2023-09-09:
> Ideally there would be a python script someone can run, providing their original format as input and it generates the SigMF files

This is trivial by using the `entry_point`s in the `setup.py`, but that kind of fully automated implementation would have to make some guesses that might be tricky.

Imagine if someone converts a 2-channel wav file. Do we assume this is I, Q or just a 2 channel real file. What about a 3 or 4 channel wav?

777arc, 2023-09-09:
I think each converter is going to have some params

KelseyCreekSoftware, 2025-12-24:
That is working much better for me under Windows 11 now!

I only have one file with real data from Sceptre. I viewed it on the IQ Engine Web site and the meta data, but not sure on the IQ content.

The other files I have from test resources for a RUST project, may be synthetic, but they all seemed to work. See screen shot from IQ Engine and a snip of debug output. 

I find the debug output useful and even had a bit of the PYSDR plotting code, to try and look at the converted signal data as a cross check.

**Sceptre Test File**

[image]

**RUST Test Files**

[image]

```console
...

DEBUG:root:parsing BLUE file data values
DEBUG:root:format: SD, data_size from header: 131072 bytes
DEBUG:root:elem_size: 8, elem_count: 16384, is_complex: False
INFO:root:wrote C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\bluefile-main\bluefile-main\resources\test\penny.prm.sigmf-data
INFO:root:Using SigMF datatype: rf64_le for BLUE format SD
WARNING:root:BLUE timecode is zero or missing; datetime metadata will be absent.
DEBUG:root:created SigMFFile({
    "global": {
        "blue:adjunct": {
            "subsize": 128,
            "xdelta": 1.0,
            "xstart": 0.0,
            "xunits": 0,
            "ydelta": 1.0,
            "ystart": 0.0,
            "yunits": 0
        },
        "blue:extended": {
            "COMMENT": "Demo data for XRTSURFACE/STAY",
            "COMMENT1": "XRTSURF/STAY/NOLAB/XC=5,PENNY,1.0,255.0,3,128,16",
            "COMMENT2": "XRTSURF/STAY/NOLAB/XC=5,PENNY,1.0,255.0,4,128,16",
            "COMMENT3": "XRTSURF/STAY/NOLAB/XC=5,PENNY,1.0,255.0,4,128,16,0,10,2",
            "COMMENT_1": "XRTSURF/STAY/NOLAB/XC=5,PENNY,1.0,255.0,4,128,16,0,10,2"
        },
        "blue:fixed": {
            "data_rep": "EEEI",
            "data_size": 131072.0,
            "data_start": 512.0,
            "detached": 0,
            "ext_size": 320,
            "ext_start": 257,
            "flagmask": 0,
            "format": "SD",
            "head_rep": "EEEI",
            "in_byte": 0.0,
            "inlet": 0,
            "keylength": 19,
            "keywords": "VER=1.1\u0000IO=X-Midas\u0000                                                                         ",
            "out_byte": 0.0,
            "outbytes": 0.0,
            "outlets": 0,
            "outmask": 0,
            "pipe": 0,
            "pipeloc": 0,
            "pipesize": 0,
            "protected": 1,
            "timecode": 0.0,
            "type": 2000,
            "version": "BLUE"
        },
        "blue:keywords": {
[... 155 lines cut ...]
DEBUG:root:outlets   : 0  # Number of outlets
DEBUG:root:outmask   : 0  # Outlet async mask
DEBUG:root:pipeloc   : 0  # Pipe location
DEBUG:root:pipesize  : 0  # Pipe size in bytes
DEBUG:root:in_byte   : 0.0  # Next input byte
DEBUG:root:out_byte  : 0.0  # Next out byte (cumulative)
DEBUG:root:outbytes  : 0.0  # Next out byte (each outlet)
DEBUG:root:keylength : 19  # Length of keyword string
DEBUG:root:keywords  : 'VER=1.1\x00IO=X-Midas\x00                                                                         '  # User defined keyword string
DEBUG:root:>>>>>>>>> User Keywords
DEBUG:root:{'VER': '1.1', 'IO': 'X-Midas'}
DEBUG:root:>>>>>>>>> Adjunct Header
DEBUG:root:{'xstart': 0.0, 'xdelta': 1.0, 'xunits': 1}
DEBUG:root:>>>>>>>>> Extended Header
DEBUG:root:B_TEST              :123
DEBUG:root:I_TEST              :1337
DEBUG:root:L_TEST              :113355
DEBUG:root:X_TEST              :987654321
DEBUG:root:F_TEST              :0.12345000356435776
DEBUG:root:D_TEST              :9.87654321
DEBUG:root:O_TEST              :/w==
DEBUG:root:STRING_TEST         :Hello World
DEBUG:root:B_TEST2             :99
DEBUG:root:STRING_TEST         :Goodbye World
```

Teque5, 2025-12-24:
Thank you for validating, glad to hear it's working correctly.

I think I only have one more thing to do before this is done, and that is make it so you can do `sigmf.fromfile()` on a `.wav` or `.cdif` or `.tmp` and it will return a SigMFFile object without writing any data back to disk. This involves creating a conforming metadatafile and using the header_bytes and trailing_bytes values. It's pretty self contained so I guess it could be a future PR also.

KelseyCreekSoftware, 2025-12-24:
I guess an "in the air" conversion without touching the disk.

We started a NonSigMF repo to store samples. If other types of 100X or 200X should be tested, and public files are available, we could store them there. https://github.com/sigmf/example_nonsigmf_recordings

It looks like the test code for the converters has been started. I was looking at Sigmffile.validate and was thinking that might be one check of a converted file.

Teque5, 2025-12-31:
@KelseyCreekSoftware please pull latest.

You should be able to do this now:
```python
>>> import sigmf
>>> meta = sigmf.fromfile('whatever.cdif')
>>> meta.get_global_field('blue:keywords')
>>> some_samples = meta.read_samples(count=128)
```

This PR is blocked by several unapproved PRs, but as of now I'm happy with how this works. Pending your feedback and those other PRs I'll create a PR for this.

KelseyCreekSoftware, 2026-01-07:
I have focused my testing on Windows command line file conversions. But looking forward to also trying the python interface to do things such as parsing converted meta data. That's pretty neat!

## Scenario *Help* and *Version*

To confirm I have pulled the latest versions I started with --help and --version

```console
PS C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\sigmf-python> python3 -m sigmf.convert.blue --version
blue.py v1.2.13
```

```console
PS C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\sigmf-python> python3 -m sigmf.convert.blue --help                                                
usage: blue.py [-h] -i INPUT [-o OUTPUT] [-v] [--archive] [--ncd] [--version]                                        

X-Midas BLUE File converter. This script reads and parses the HCB (Header Control Block) and Extended Headers. It supports different file types and
extracts metadata accordingly. Converts the extracted metadata into SigMF format.

options:
  -h, --help            show this help message and exit
  -i INPUT, --input INPUT
                        BLUE file path
  -o OUTPUT, --output OUTPUT
                        SigMF path
  -v, --verbose
  --archive             Write a .sigmf archive instead of meta/data pair
  --ncd                 Process as Non-Conforming Dataset and write .sigmf-meta only.
  --version             show program's version number and exit
```

Perhaps we could mention that 1000, 1001 and 2000 file types are supported in file comment on line 12 of blue.py? 

We could also mention the nifty handling of duplicate tags, by post appending a count to them.

When the code is live, we could work with Mark to add something like a “HowTo” to the IQ. Perhaps that is the place for additional detail.

If my understanding is correct, it would not make sense to specify --ncd and --archive file. So perhaps use a mutually_exclusive_group for those two?

```python
group = parser.add_mutually_exclusive_group()
group.add_argument("--archive", action="store_true", help="Write a .sigmf archive instead of meta/data pair")
group.add_argument("--ncd", action="store_true", help="Process as Non-Conforming Dataset and write .sigmf-meta only.")
```

## Scenario *--i* and *--o*

(On all of these test scenarios I also add *--verbose*)

```console
PS C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\sigmf-python> python3 -m sigmf.convert.blue --i C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\SceptreTestFile1.cdif --o C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\testbluefiles --verbose
INFO:root:Read BLUE type 1001 using Platinum specification.
INFO:root:wrote C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\testbluefiles.sigmf-data
INFO:root:Using SigMF datatype: cf32_le for BLUE format CF
INFO:root:wrote C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\testbluefiles.sigmf-meta
```

This works OK, but instead of using `C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\testbluefiles\' as the output directory, it writes two files named 'testbluefiles.*'. 

Similar result with a trailing slash.

```console
PS C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\sigmf-python> python3 -m sigmf.convert.blue --i C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\SceptreTestFile1.cdif --o C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\testbluefiles\ --verbose
INFO:root:Read BLUE type 1001 using Platinum specification.
INFO:root:wrote C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\testbluefiles.sigmf-data
INFO:root:Using SigMF datatype: cf32_le for BLUE format CF
INFO:root:wrote C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\testbluefiles.sigmf-meta
```

## Scenario *--ncd*

[... 40 lines cut ...]
    self.cleanup()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2544.0_x64__qbz5n2kfra8p0\Lib\tempfile.py", line 947, in cleanup
    self._rmtree(self.name, ignore_errors=self._ignore_cleanup_errors)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2544.0_x64__qbz5n2kfra8p0\Lib\tempfile.py", line 929, in _rmtree
    _shutil.rmtree(name, onerror=onerror)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2544.0_x64__qbz5n2kfra8p0\Lib\shutil.py", line 787, in rmtree
    return _rmtree_unsafe(path, onerror)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2544.0_x64__qbz5n2kfra8p0\Lib\shutil.py", line 634, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.2544.0_x64__qbz5n2kfra8p0\Lib\tempfile.py", line 893, in onerror
    _os.unlink(path)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\KE7AR\\AppData\\Local\\Temp\\tmpj21aitgz\\testbluefiles.sigmf-data'
```

## Misc Comments

Line 277 of blue.py – Perhaps add a log.info entry for use of S1 for unknown types?

On line 731:

    out_path : str, optional
        Path to the output SigMF metadata file.

I would consider adding the comment from line 744 - *auto-enable NCD when no output path is specified.*

KelseyCreekSoftware, 2026-01-14:
Looks great, everything is working very well!

## Scenario Help and Version

To confirm I have pulled the latest versions I started with --help and --version

### Version

```console
PS C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\sigmf-python> python3 -m sigmf.convert --version
__main__.py v1.6.0
```

### Help

Calling the sigmf.convert' without any parameters returns the command line usage help. 

```console
PS C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\sigmf-python> python3 -m sigmf.convert                                                      
usage: sigmf_convert [-h] [-v] [-a | --ncd] [--version] input output
```

The expanded full --help should be super helpful for those writing automation for this CLI!

```console
PS C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\sigmf-python> python3 -m sigmf.convert --help
usage: sigmf_convert [-h] [-v] [-a | --ncd] [--version] input output
Unified entry-point for SigMF conversion of non-SigMF recordings.

This command-line interface converts various non-SigMF file formats into SigMF-compliant datasets.
It currently supports WAV and BLUE/Platinum file formats.
The converter detects the file type based on magic bytes and invokes the appropriate conversion function.

By default it will output a SigMF pair (.sigmf-meta and .sigmf-data).

Converter Processing Pattern
----------------------------
if out_path is None:
    create_ncd = True
<create global_info and capture_info>
if create_ncd:
    <create Non-Conforming Dataset (NCD) with .sigmf-meta only>
    if out_path:
        <write out_path.sigmf-meta>
    return SigMFFile
if create_archive:
    with TemporaryDirectory() as temp_dir:
        <write .sigmf-data>
    <write out_path.sigmf>
else:
    <write out_path.sigmf-data>
    <write out_path.sigmf-meta>
return SigMFFile

positional arguments:
  input          Input recording path
  output         Output SigMF path (no extension)

options:
  -h, --help     show this help message and exit
  -v, --verbose  Increase verbosity level
  -a, --archive  Output .sigmf archive only
  --ncd          Output .sigmf-meta only and process as a Non-Conforming Dataset (NCD)
  --version      show program's version number and exit
```

## Scenario Required Input and Output file without additional parameters

I used a .tmp and .cdif input file and they both worked great!

[... 62 lines cut ...]
### Attempting to specify both --ncd and --archive.

Appropriately Returned `argument -a/--archive: not allowed with argument --ncd` when attempting to specify both --ncd and --archive.

```console
PS C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\sigmf-python> python3 -m sigmf.convert C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\SceptreTestFile1.cdif C:\Data1\Ham_Radio\SDR\SigMF-MIDAS-Blue-File-Conversion\testbluefiles --ncd --archive --verbose
usage: sigmf_convert [-h] [-v] [-a | --ncd] [--version] input output
sigmf_convert: error: argument -a/--archive: not allowed with argument --ncd
```

## Misc. Comments

### IQ File converted validity

Other than giving a few converted files a quick look in IQEngine, I have not been able to validate the converted IQ data files. Longer term it could be handy to have something like 1 kHz test tone files, or some broadcast FM spectrum samples as a sanity check for converted files.

Personally, I think synthetic files are fraught, but that might be an adjunct for IQ conversion validation automation to augment real world recorded files.

### Python interface

Still hoping to test the python interface for at least file conversion. 

### Future Kraken SDR conversion

Might be nice to consider Kraken SDR files for conversion as well.