sigmf/sigmf-python #26: Support for compressed archives

Teque5, 2023-09-08:
Back when this repo was the spec, we considered it out-of-scope to handle the compression side, but I think if we wanted we could support `.sigmf.7z` and `.sigmf.zip` files with this module. If anyone thinks this is a good idea I can work up a PR.

aromanielloNTIA, 2023-10-12:
This would be an excellent addition!

Teque5, 2025-01-07:
For my own reference this can easily be done with our existing implementation using `tarfile.TarFile` which should allow `x.sigmf.gz`, `x.sigmf.bz2`, and `x.sigmf.xz`.

To allow `x.sigmf.zip` I believe we would need to use the similar `zipfile.ZipFile` object.

To allow `x.sigmf.7z` I see there is `py7zr.SevenZipFile` although I need to investigate further.