sigmf/sigmf-python #36: How to implement the spatial extension

MaxwellMorris, 2023-11-01:
Hello

I would like to be able to work with the spatial extension with this library:
https://github.com/sigmf/SigMF/blob/sigmf-v1.x/extensions/spatial.sigmf-ext.md

I am currently a little bit confused as to how an SigMF extension can integrated with this library.

Would it be possible for a maintainer to:

A. Outline how to extend the metadata with an extension in the current state.
B. If that capability doesn't exist briefly describe a process that they would like to be implemented to add metadata extension and I will attempt to make a pull request.

I apologize if I have missed something.

Thank you for your attention.

Cheers,
Max

MaxwellMorris, 2023-11-01:
https://github.com/sigmf/sigmf-python/issues/6#issuecomment-1781465706
I will use this as a work around in the mean time.

Teque5, 2024-12-20:
The extensions have been moved to [their own repository](https://github.com/sigmf/community-extensions), but yes we should probably have an example of how to implement an extension.