sigmf/sigmf-python #41: Example of usage of Extensions Namespaces in sigmf-python

bhorsfield, 2023-12-22:
Hi All,

Can anyone provide me with a simple Python code snippet that demonstrates the usage of the **Extension Namespaces** feature in SigMF?

I have searched the Issues section in this repo, and performed a general internet search, but I cannot find any examples of this feature being used. 

In particular, I am interested in examples of the following functionality:

1. Use built-in SigMF extensions (e.g. antenna, signal, spatial)
2. Create custom extensions
3. Validation of contents of metadata objects to ensure that extensions fields have been defined correctly (e.g. correct data type, correct spelling of field names...). 

Note that I seem to have no trouble adding completely arbitrary fields to my sigmf-meta file and reading them back in again.  For example:

meta.add_capture(0, metadata={
    SigMFFile.FREQUENCY_KEY: 915000000,
    SigMFFile.DATETIME_KEY: dt.datetime.utcnow().isoformat()+'Z',
    SigMFFile.LAT_KEY: -90.0,
    SigMFFile.LON_KEY: 157.66,
    SigMFFile.LENGTH_INDEX_KEY: 122880,
    "my_extension:latitude": 111.0,
    "my_extension:longitude": 22.0,
    "my_extension:num_samples": 122880
})

However, this is not a great solution, as it does not force the user to conform to a pre-defined template or schema.

Any assistance or advice you can provide would be greatly appreciated!

Thanks,
Brendan.

Teque5, 2024-01-02:
Extensions can be will defined additions to the metadata ([see NTIA examples](https://github.com/NTIA/sigmf-ns-ntia)), but this python module has no way of automatically validating those extensions unless a schema is defined for that extension. 

[There is a proposal to generate the specification directly from the schema](https://github.com/sigmf/SigMF/pull/292), but due to the added complexity I think we are afraid to merge that PR without more community input.

When I have custom fields to add I typically do so similar to your example.

bhorsfield, 2024-01-03:
Thanks Teque5.  

I had assumed it was **_mandatory_** to validate my `sigmf-meta` files against a schema when using an extension.  However, it now sounds like this is really my decision to make, depending on how rigorously I want to enforce the structure of my metadata files.

This is a relief, because JSON schemas look very labour intensive to create & maintain!

Teque5, 2024-01-03:
> This is a relief, because JSON schemas look very labour intensive to create & maintain!

Yes I wish there was a better way. The syntax is adequate but not human-readable. On another project I use jsonschema but keep the actual schemas as a yaml -> which is obv better but not typical. If there was an OSS GUI for jsonschema more people would use it.

closed 2024-01-03 by Teque5