sigmf/sigmf-python #46: Allow skip_checksum on creation

Teque5, 2024-01-13:
As mentioned [in the other repo](https://github.com/sigmf/SigMF/issues/304#issuecomment-1890243814), the checksum field is optional yet this repo always calculates when creating a new dataset. Need a simple PR to allow skip. Probably combine with #44.

abhaysamantni, 2024-01-14:
Thank you for raising the issue here @Teque5 . What is the tentative timeline for fixing the issue?

Teque5, 2024-09-10:
After looking into this, I realized it's already possible to skip checksum on creation. 

An example similar to the current `README.md`, note I only added `skip_checksum=True`:
```python
from sigmf import SigMFFile
import numpy as np

data = np.zeros(1024, dtype=np.complex64)
data.tofile('example_cf32.sigmf-data')
meta = SigMFFile(
    data_file='example_cf32.sigmf-data',
    global_info = {
        SigMFFile.DATATYPE_KEY: 'cf32_le',
        SigMFFile.SAMPLE_RATE_KEY: 48000,
        SigMFFile.VERSION_KEY: '1.2.0',
    },
    skip_checksum=True,
)
```
and the resulting object will have no checksum on creation:
```python
>>> meta
SigMFFile({
    "global": {
        "core:datatype": "cf32_le",
        "core:sample_rate": 48000,
        "core:version": "1.2.0"
    },
    "captures": [],
    "annotations": []
})
```
and you can still calculate the checksum manually with `meta.calculate_hash()`, resulting in:
```python
>>> meta
SigMFFile({
    "global": {
        "core:datatype": "cf32_le",
        "core:sample_rate": 48000,
        "core:sha512": "18790c279e0ca614c2b57a215fecc23a6c3d2d308ce77f314378cb2d1b0f413acd3a9cd353aa6da86ec9f51916925c7210f7dfabc0ef726779f8d44f227f03b1",
        "core:version": "1.2.0"
    },
    "captures": [],
    "annotations": []
})
```
and the metadata validates correctly either way.

closed 2024-09-10 by Teque5