sigmf/sigmf-python #62: Unable to run sigmf_gui

bhorsfield, 2024-07-04:
Hi Folks,

I have just tried to run the utility **sigmf_gui** for the first time, but each time I try an error dialog pops up with the message "Unable to complete operation on element with key Update":

[image]

As shown, the error dialog offers a suggested fix for the problem, but this has made no difference:

`window = Window('SigMF Archive Creator', auto_size_buttons=False, default_element_size=(20, 1), auto_size_text=False, default_button_element_size=(10, 1), finalize=True).Layout(layout)
`

Details of my system configuration are as follows:
- HP Omen laptop with i7 processor & 32 GB of RAM
- Ubuntu 22.04
- Python 3.10.12
- SigMF version 1.2.1 

Any tips or suggestions as to the root cause of this problem would be greatly appreciated.

Thanks & Regards,
Brendan.

gregparkes, 2024-07-26:
Hi Brendan,

I've encountered this issue also. Unfortunately PySimpleGUI is now no longer open-source as of v5.0 and has gone commercial (hence that menacing TRIAL PERIOD message at the bottom). 

A quick fix would be for the sigmf-python maintainers to use the FreeSimpleGUI package which is a copy of the v4.* PySimpleGUI under its old open-source license.

https://github.com/spyoungtech/FreeSimpleGui

777arc, 2024-07-26:
We've actually been considering removing the GUI portion of this library, e.g. moving it to a different archived repo in case one day someone wants to start it back up and maintain it, because we haven't had a maintainer for that portion of code in a while.

bhorsfield, 2024-08-21:
Understood, thanks for letting me know.

PR #66 (2024-09-10): remove sigmf_gui due to lack of support and PySimpleGUI changes
Closes #62.
closed 2024-08-21 by bhorsfield