sigmf/sigmf-python #88: building with setuptools 75.2.0-2 gives a warning about sigmf.apps

chrisjohgorman, 2024-12-18:
When building `python-sigmf` under Arch Linux (`python 3.12.7-1` and `python-setuptools 1:75.2.0-2`) I get a warning about `sigmf.apps` being missing from the packages configuration.

```
/usr/lib/python3.12/site-packages/setuptools/command/build_py.py:218: _Warning: Package 'sigmf.apps' is absent from the `packages` configuration.
!!

        ********************************************************************************
        ############################
        # Package would be ignored #
        ############################
        Python recognizes 'sigmf.apps' as an importable package[^1],
        but it is absent from setuptools' `packages` configuration.

        This leads to an ambiguous overall configuration. If you want to distribute this
        package, please make sure that 'sigmf.apps' is explicitly added
        to the `packages` configuration field.

        Alternatively, you can also rely on setuptools' discovery methods
        (for example by using `find_namespace_packages(...)`/`find_namespace:`
        instead of `find_packages(...)`/`find:`).

        You can read more about "package discovery" on setuptools documentation page:

        - https://setuptools.pypa.io/en/latest/userguide/package_discovery.html

        If you don't want 'sigmf.apps' to be distributed and are
        already explicitly excluding 'sigmf.apps' via
        `find_namespace_packages(...)/find_namespace` or `find_packages(...)/find`,
        you can try to use `exclude_package_data`, or `include-package-data=False` in
        combination with a more fine grained `package-data` configuration.

        You can read more about "package data files" on setuptools documentation page:

        - https://setuptools.pypa.io/en/latest/userguide/datafiles.html

        [^1]: For Python, any directory (with suitable naming) can be imported,
              even if it does not contain any `.py` files.
              On the other hand, currently there is no concept of package data
              directory, all directories are treated like packages.
        ********************************************************************************

!!
```
The attached patch adds `sigmf.apps` to the packages list and makes this go away, but I wanted to log it here in case my understanding of the error and how to fix it is wrong.

[0001-packages-sigmf.apps.patch.txt](https://github.com/user-attachments/files/18185510/0001-packages-sigmf.apps.patch.txt)

Teque5, 2024-12-18:
Thanks. I will test this an make a PR later this week.

Teque5, 2024-12-19:
I'm unable to replicate this with python 3.12.7 and setuptools 75.6.0 in a docker container. Note within the source code for that file it references [setuptools issue 3340](https://github.com/pypa/setuptools/issues/3340) which seems particularly controversial.

What command were you running?

I am able to run these without that warning:
* `pip install .`
* `python3 -m build --wheel`
* `python3 -m build --sdist`

chrisjohgorman, 2024-12-19:
Hi @Teque5, I am running `python3 -m build --wheel --no-isolation`.  I assume I'm getting this because I am running the command on my daily work machine, which has lots of python programs installed and is not sanitary like your docker container.  Thanks for pointing me to the code and setuptools issue 3340.  Apologies for the make work project.  I'll close the issue.

closed 2024-12-19 by chrisjohgorman

Teque5, 2024-12-19:
No big deal, I'm happy to fix any other issues you find.

For the record `python3 -m build --wheel --no-isolation` seems to work okay at the moment. If you can replicate it somehow please reopen the issue.