sigmf/sigmf-python #89: OS independence

GreenK173, 2024-12-19:
According to `pyproject.toml`,  this library should be OS independent. However I've found two cases where the code only works on Unix platforms but not on Windows.  One was during discussion of PR #81. The other one is the use of [os.sched_getaffinity](https://docs.python.org/3/library/os.html#os.sched_getaffinity) in `validate.py`, which is only available on Unix (as mentioned at the beggining of the section in the doc).

Teque5, 2024-12-19:
Thanks for noting this issue with windows. We can either leave a note where we manually garbage collect (specifically for this windows issue) or we can find a function that can remove the memmap file.

I wonder, does `pathlib.Path.unlink()` work for this problem? Generally I think we should have all our paths in this module handled by pathlib anyway, but it's just been on the to-do list. In unix `unlink()` will either remove the file or the softlink.

GreenK173, 2025-01-08:
```pathlib.Path.unlink()``` doesn't help, it results in the same ```PermissionError```. I'm not an expert but I think in Windows it just won't allow you to delete a file used by a different process. And the [NumPy doc](https://numpy.org/doc/stable/reference/generated/numpy.memmap.html) says:
>  Currently there is no API to close the underlying ```mmap```. It is tricky to ensure the resource is actually closed, since it may be shared between different memmap instances.

So garbage collection is the only way I see.

PR #101 (2025-01-15): cross-platform worker estimation
Closes #89.
closed 2025-01-15 by Teque5