sigmf/sigmf-python #92: move keys from sigmf.SigMFFile to top-level sigmf

Teque5, 2024-12-19:
It's bothered me ever since the beginning that if you want to set or get some key within `sigmf` you needed to do `sigmf.SigMFFile.VERSION_KEY` instead of just doing `sigmf.VERSION_KEY`. Now we appear to also have keys within `archive.py` for file extensions, which is strange.

Other modules (`logging`, `pygame`, `numpy`) have global vars like this in the top level.

I would like to:

1. Consolidate all keys into `keys.py` or similar name.
2. In the `__init__.py` do `from .keys import *` so that we can access a key using `sigmf.VERSION_KEY`.
3. In `SigMFFile` attach keys as properties for now and insert a comment saying that functionality will be deprecated in a future version. This keeps backwards compatibility.

This change should increment the minor version (`major.minor.patch`).

GreenK173, 2025-01-08:
Agree on all points, especially the file extension keys being in `archive.py` makes little sense. Also noticed that some of the code doesn't actually use these keys and uses the hard-coded extensions instead.

In `archivereader.py`:

```python
raise SigMFFileError("No .sigmf-data file found in archive!")
```

In `sigmffile.py`:

```python
if metafile.endswith(".sigmf-meta") and path.isfile(metafile_path):
```

```python
metafile = stream_name + ".sigmf_meta"
```

```python
metafile = self.get_stream_names()[stream_index] + ".sigmf_meta"
```

The last two are actually incorrect (`.sigmf_meta` instead of `.sigmf-meta`) ! However this bug doesn't manifest itself since `fromfile` treats every extension other that `.sigmf` and `.sigmf-collection` as if it were `.sigmf-meta` (perhaps that's also not ideal).

Teque5, 2026-09-21:
Closed by #150

closed 2026-09-21 by Teque5